MinnowBoard-org/bugs-and-help #79: The Website MinnowBoard.org has issues.

avramoff, 2017-11-21:
My freshly installed and perfectly tuned Chromium browser, Version 62.0.3202.89 (Official Build) Built on Ubuntu, running on LinuxMint 18.2 (64-bit) is scrolling websites and webapps as a Google Chrome Stable on a Chromebook!... Except on the new (21st Nov 2017) website of MinnowBoards.org - the scrolling up and down across all the pages of the site, constantly and very unpleasantly jumps up and down, and practically it's hard to fix a page exactly where you want. 

I'd tried it with Google Chrome Version 62.0.3202.94 (Official Build) (64-bit), downloaded and installed directly from Google.com/Chrome on a very stable Linux Mint 18.2 based on Ubuntu 16.04 with ntp, nodejs, angularjs, and Google's V8 JavaScript, integrated with the Chromium.org browsers - and the result is the same. 

Then I'd tried it with the new Mozilla Firefox 57.0 (Quantum), and - the same issue.

My network on that Linux Mint 18.2 is running only through highly hardened encryption by an original up-to-date OpenVPN.Net client, but this issue with scrolling the pages, that pre-jump-like up and down, I found only on MinnowBoard.org, which I found on 01.org ... I respect Intel's enthusiasm and solidarity with the OSS and FOSS GNU/Linux Global Community.

Conclusion: The new, totally rebuilt website MinnowBoard.org has issues while scrolling up and down with Chromium.org based browsers, including the Stabe Google Chrome.

gvancuts, 2017-11-22:
Thanks for the report @avramoff . I don't have the exact same set-up as yours but I am unable to see these problems on a Fedora 26 with Google Chrome Version 62.0.3202.75 (Official Build) (64-bit). I will update Google Chrome just now to see if it makes any difference.

gvancuts, 2017-11-22:
Just tried with Google Chrome  Version 62.0.3202.94 (Official Build) (64-bit) and also tried on Windows 10 Enterprise using Google Chrome Version 62.0.3202.94 (Official Build) (64-bit). I did not notice any real issue (although some of the pictures were a little slow to load here).

Could this be related to pages loading slowly? Or does it look more like graphical artefacts (in which case, your exact HW configuration may help root-cause the issue). Does that seem triggered by the banners at the top of the page that move a bit when scrolling up and down (especially the top one which sometimes hides away and causes some movement of the rest of the page content)?

Mark-van-der-Pol, 2018-01-22:
There are some lingering minor visual anomalies scrolling the website on some browsers, but these don't materially interfere with the usability of the website.  And these don't occur to me as "constantly and very unpleasantly jumps up and down,.. "

I am closing this issue as it seems the artifact mentioned has been ameliorated in a site or browser update, and is not showing up as described anymore.

|\\/|ark.

closed 2018-01-22 by Mark-van-der-Pol