MinnowBoard-org/bugs-and-help #80: Windows IoT Core instructions error64-bit

chsha, 2017-11-23:
On https://minnowboard.org/tutorials/installing-windows-iot-core the directions say to update the firmware for the MBM/Turbot to 32-bit to install IoT Core. This is no longer necessary. The FFU provided by Microsoft is now 64-bit. Please correct the tutorial.

uhofemei, 2017-11-27:
Hi Chirag,

I forwarded this information to the owner of the website tutorial. Thank  you for your feedback :)

Ulf

veyalla, 2018-01-22:
@uhofemei - any update? The documentation is not fixed yet.

uhofemei, 2018-01-22:
@veyalla - it appears the ball on this one got dropped. I contacted the person in charge of the tutorial and it should be fixed by the end of this week. Sorry about that!

keerockl, 2018-01-26:
@veyalla -Tutorial has been modified. Thank you for reporting!

closed 2018-01-26 by keerockl