MinnowBoard-org/bugs-and-help #81: I2C write error on Minnow Board Turbot (Dual Ethernet)

gvamsk, 2017-12-01:
Hi ,

I have connected SDA,SCL and GND pins between MinnowBoard and my sensor, I am able to open i2c device entry but the first write itself fails with errno 6. 

I am not able to write over I2C (any of the channels 0,1,5). I always get write error: 6. 

Can you let us know the details about Device entry(0,1,2..9) to be used from my application when using I2C @pin 13 and 15 on Device.

Thanks,
Vamsi

gvancuts, 2017-12-01:
Based on [this page](https://github.com/intel-iot-devkit/mraa/blob/master/docs/minnow_max.md#interface-notes) you should be using `/dev/i2c-7`. Is this already the device you are using?

gvamsk, 2017-12-01:
Yes, I have checked i2c-7 too. It's same error.

gvancuts, 2017-12-01:
> Yes, I have checked i2c-7 too. It's same error.
Thanks for confirming this. Can you share your code snippet? What's your base OS?

Mark-van-der-Pol, 2017-12-01:
Unfortunately, while it is bus 7, it is device 5 for obscure reasons in kernel land. And the MRAA obfuscates matters further by referring to the buses by how many MRAA controls or knows about. So the first i2c bus is referred to as 0, the second one as 1, etc.

So in MRAA world, talk to i2c instance #0, and it is mapped onto the pins that are on the LSE on the Turbot.

And all this is from memory, so it may not reflect reality.

|\/|ark.

Mark-van-der-Pol, 2018-01-22:
I have tried I2C on some Dual Ethernet boards, and I get correct behaviour. The bus numbers are a bit confusing, with i2cdetect needing bus 0 or 8 depending on ... well, not sure, actually.

But the UPM examples programs for the sensors I tried were happy.

I am closing this, but if you still have issues, please re-open and provide more details.

|\\/|ark

closed 2018-01-22 by Mark-van-der-Pol