MinnowBoard-org/bugs-and-help #82: Problem turning on a Minnowboard Quad Core

tranfa, 2017-12-09:
Good morning, 

I'm trying to turn on my Minnowboard Quad Core Turbot B for the first time, and I struggle.

I find myself in the situation described here at point 2:
https://minnowboard.org/tutorials/getting-started-with-minnowboard-turbot
The D1 led and the fan turn on, but the D2 doesn't, and the HDMI connection to the monitor doesn't give any sign of life - so I presume the CPU isn't on.

The last of the FAQ here 
https://minnowboard.org/support
says that "If the D2 light does not come on at all, this can mean the board has been damaged by using a power supply greater than 5V", but I'm using a 5V linear HDPlex tested, controlled and used in other instances without problems - leaving aside that this is a Turbot B, which has over-voltage protection on board.

Tried the reset button, but nothing changes.

Could anyone give me some guidance?

Thanks in advance!

Tranfa

tranfa, 2017-12-12:
Solved, was due to a faulty connector....

Mark-van-der-Pol, 2017-12-12:
Faulty connector where?
If the D1 lights, there is power at the board. If D2 doesn't light, and the power/reset button doesn't turn it on, it is strange. Unless the faulty connector is in the HDMI cable, draining too much power, or if you had something plugged in to the USB ports draining too much power.

Please share where the faulty connector played a part. 

Thanks.
|\\/|ark.

tranfa, 2017-12-13:
It was the connector at the termination of the PS cable to be faulty, so probably not bringing enough power to the board.
I just substituted it, and everything started to work fine.

Mark-van-der-Pol, 2017-12-13:
Great, thank you for letting me know. I will put a troubleshooting guide on the MinnowBoard.org website that will incorporate this and some other subtle issues that might come up.
|\\/|ark.

closed 2017-12-13 by Mark-van-der-Pol

naiyining, 2019-05-08:
I have the same problem.I have only one M.2 SATA disk.how to change the ic to draining too much power?
Thanks.