MinnowBoard-org/bugs-and-help #86: Minnowboard Max Derivative with DisplayPort Support

dscully-sbis, 2018-01-10:
Hello,

I have a custom platform that was based off the Minnowboard Max design. Since most of the changes are downstream of the SoC (e.g., adding more USB devices on board), I was able to load the stock firmware from the Intel firmware [website](https://firmware.intel.com/projects/minnowboard-max) without having to rebuild the firmware myself.

I would now like to spin the design to hook up the DisplayPort lines for a secondary output along with the existing HDMI port. I'm aware that the DisplayPort output on DDI1 is not hooked up on the Minnowboard Max. My question is are firmware changes required to enable this interface? If so, could someone point me in the right direction on what needs to be modified in the edk2 source tree in order to enable this interface?

Thanks,
David

Mark-van-der-Pol, 2018-04-15:
Hmm, for sure the Display Port supports higher resolutions than that. Not sure what to suggest in that regard - could be just a Yocto thing. 
As a diagnostic step, you might try putting Ubuntu on the board to see what it sees. Both in terms of dual display, as well as higher resolution.
|\\/|ark.

BrianRichardsonIntel, 2018-04-16:
Adding an additional video output like DisplayPort requires a modification to the configuration file used by the Intel Graphics Output Protocol (GOP) driver for proper initialization. The Intel BMP utility is not available in open source, but Intel customers can obtain it from their sales/field representative.

dscully-sbis, 2018-04-16:
Thank you both for getting back to me. I was able to obtain a copy of the BMP utility from the https://platformsw.intel.com/ website.

I would like to confirm the files/settings I will need to edit to enable the DisplayPort:

It appears that the PlatformPkgGcc points to this file for the VBT:
Vlv2MiscBinariesPkg/GOP/7.2.1011/VBT/MNW2/Vbt.bin

My plan is to change "EFP 2" Device Type under Display Configuration to "DisplayPort" in the BMP tool and recompile.

I'll update this card with my results after test.

rksaripalli, 2019-05-30:
Hi, I have a similar need for a board we have designed based on the Baytrail e3800 SoC. I am using the Minnowboard UEFI FW as well. I have a custom LVDS panel on the eDP1 interface and HDMI. HDMI works fine and I need to enable LVDS panel. The LVDS panel has EDID populated and it is connected through the DDC channel (DDI1) to the SoC.

It looks like I am trying to do exactly the same as yours. 

Did your steps work?. Can you confirm?. If you had to modify any of the steps above, could you let me know please?

Thanks,
RK

dscully-sbis, 2019-05-30:
Hey,

Sorry we abandoned this project shortly after posting this. I believe that I was able to make the modification with the BMP tool and build the test firmware. I don't recall if it worked however.

Sorry and good luck!

rksaripalli, 2019-05-30:
Thank you very much for your quick update. Appreciate the response

closed 2019-05-30 by dscully-sbis