MinnowBoard-org/bugs-and-help #92: MinnowBoard UDK2017 build fails

pietrushnic, 2018-02-14:
I'm following [Release Notes](https://firmware.intel.com/sites/default/files/MinnowBoard_MAX-Rel_0_97-ReleaseNotes.txt) using Linux (Debian sid), but build fails with below error. I have correct `GCC5` toolchain - it looks to me like bug in code, but I assume someone tested that before and it had to work. Any idea what can be wrong ?

```shell
/home/edk2/edk2-platforms/Vlv2TbltDevicePkg/Library/MultiPlatformLib/PlatformInfoHob.c: In function 'GetPlatformInfoHob':                                                                                                            
/home/edk2/edk2/MdePkg/Include/Base.h:951:75: error: comparison of constant '0' with boolean expression is always false [-Werror=bool-compare]                                                                                        
 #define RETURN_ERROR(StatusCode)     (((INTN)(RETURN_STATUS)(StatusCode)) < 0)                                                                                                                                                      
                                                                           ^                                                                                                                                                     
/home/edk2/edk2/MdePkg/Include/Uefi/UefiBaseType.h:165:35: note: in expansion of macro 'RETURN_ERROR'                                                                                                               
 #define EFI_ERROR(A)              RETURN_ERROR(A)                                                                                                                                                       
                                   ^~~~~~~~~~~~      
/home/edk2/edk2/MdePkg/Include/Library/DebugLib.h:341:13: note: in expansion of macro 'EFI_ERROR'
         if (EFI_ERROR (StatusParameter)) {                                               \
             ^~~~~~~~~
/home/edk2/edk2-platforms/Vlv2TbltDevicePkg/Library/MultiPlatformLib/PlatformInfoHob.c:61:3: note: in expansion of macro 'ASSERT_EFI_ERROR'
cc1: all warnings being treated as errors
make: *** [GNUmakefile:647: /home/edk2/Build/Vlv2TbltDevicePkg/DEBUG_GCC5/IA32/Vlv2TbltDevicePkg/Library/MultiPlatformLib/MultiPlatformLib/OUTPUT/PlatformInfoHob.obj] Error 1

build.py...
 : error 7000: Failed to execute command
        make tbuild [/home/edk2/Build/Vlv2TbltDevicePkg/DEBUG_GCC5/IA32/Vlv2TbltDevicePkg/Library/MultiPlatformLib/MultiPlatformLib]

build.py...
 : error F002: Failed to build module
        /home/edk2/edk2-platforms/Vlv2TbltDevicePkg/Library/MultiPlatformLib/MultiPlatformLib.inf [IA32, GCC5, DEBUG]

- Failed -
Build end time: 15:09:48, Feb.14 2018
Build total time: 00:00:58

/home/edk2/edk2-platforms/Vlv2TbltDevicePkg/Stitch/IFWIHeader/IFWI_HEADER.bin
Skip Running fce...
Skip Running KeyEnroll...
cp: cannot stat '/home/edk2/Build/Vlv2TbltDevicePkg/DEBUG_GCC5/FV/VLV.fd': No such file or directory
cat: /home/edk2/MNW2MAX_X64_D_0097_01.ROM: No such file or directory
Skip Running BIOS_Signing ...

Build location: Build/Vlv2TbltDevicePkg/DEBUG_GCC5
BIOS ROM Created: MNW2MAX_X64_D_0097_01.ROM
```

pietrushnic, 2018-02-14:
Ok, I was able to workaround issue with this patch:

```patch
diff --git a/Vlv2TbltDevicePkg/Library/MultiPlatformLib/MultiPlatformLib.inf b/Vlv2TbltDevicePkg/Library/MultiPlatformLib/MultiPlatformLib.inf
index 741abe241063..b40e3a467873 100644
--- a/Vlv2TbltDevicePkg/Library/MultiPlatformLib/MultiPlatformLib.inf
+++ b/Vlv2TbltDevicePkg/Library/MultiPlatformLib/MultiPlatformLib.inf
@@ -82,3 +82,6 @@
 
 [Guids]
   gEfiSetupVariableGuid
+
+[BuildOptions]
+  GCC:*_*_*_CC_FLAGS = -Werror=bool-compare
```

Now facing different problem:

```
"gcc" -o /home/edk2/Build/Vlv2TbltDevicePkg/RELEASE_GCC5/IA32/Vlv2TbltDevicePkg/Feature/Capsule/SystemFirmwareDescriptor/SystemFirmwareDescriptor/OUTPUT/./SystemFirmwareDescriptor.dll -nostdlib -Wl,-n,-q,--gc-sections -z common-page-size=0x40 -Wl,--entry,ReferenceAcpiTable -u ReferenceAcpiTable -Wl,-m,elf_i386 /home/edk2/Build/Vlv2TbltDevicePkg/RELEASE_GCC5/IA32/Vlv2TbltDevicePkg/Feature/Capsule/SystemFirmwareDescriptor/SystemFirmwareDescriptor/OUTPUT/./SystemFirmwareDescriptor.obj
"GenFw" -o /home/edk2/Build/Vlv2TbltDevicePkg/RELEASE_GCC5/IA32/Vlv2TbltDevicePkg/Feature/Capsule/SystemFirmwareDescriptor/SystemFirmwareDescriptor/OUTPUT/./SystemFirmwareDescriptor.acpi -c /home/edk2/Build/Vlv2TbltDevicePkg/RELEASE_GCC5/IA32/Vlv2TbltDevicePkg/Feature/Capsule/SystemFirmwareDescriptor/SystemFirmwareDescriptor/OUTPUT/./SystemFirmwareDescriptor.dll
GenFw: ERROR 3000: Invalid
make: *** [GNUmakefile:423: /home/edk2/Build/Vlv2TbltDevicePkg/RELEASE_GCC5/IA32/Vlv2TbltDevicePkg/Feature/Capsule/SystemFirmwareDescriptor/SystemFirmwareDescriptor/OUTPUT/SystemFirmwareDescriptor.acpi] Error 2
  /home/edk2/Build/Vlv2TbltDevicePkg/RELEASE_GCC5/IA32/Vlv2TbltDevicePkg/Feature/Capsule/SystemFirmwareDescriptor/SystemFirmwareDescriptor/OUTPUT/./SystemFirmwareDescriptor.dll unsupported ELF EM_386 relocation 0xa.
GenFw: ERROR 3000: Invalid
  /home/edk2/Build/Vlv2TbltDevicePkg/RELEASE_GCC5/IA32/Vlv2TbltDevicePkg/Feature/Capsule/SystemFirmwareDescriptor/SystemFirmwareDescriptor/OUTPUT/./SystemFirmwareDescriptor.dll unsupported ELF EM_386 relocation 0x9.
GenFw: ERROR 3000: Invalid
  /home/edk2/Build/Vlv2TbltDevicePkg/RELEASE_GCC5/IA32/Vlv2TbltDevicePkg/Feature/Capsule/SystemFirmwareDescriptor/SystemFirmwareDescriptor/OUTPUT/./SystemFirmwareDescriptor.dll unsupported ELF EM_386 relocation 0xa.
GenFw: ERROR 3000: Invalid
  /home/edk2/Build/Vlv2TbltDevicePkg/RELEASE_GCC5/IA32/Vlv2TbltDevicePkg/Feature/Capsule/SystemFirmwareDescriptor/SystemFirmwareDescriptor/OUTPUT/./SystemFirmwareDescriptor.dll unsupported ELF EM_386 relocation 0x9.

build.py...
 : error 7000: Failed to execute command
        make tbuild [/home/edk2/Build/Vlv2TbltDevicePkg/RELEASE_GCC5/IA32/Vlv2TbltDevicePkg/Feature/Capsule/SystemFirmwareDescriptor/SystemFirmwareDescriptor]

build.py...
 : error F002: Failed to build module
        /home/edk2/edk2-platforms/Vlv2TbltDevicePkg/Feature/Capsule/SystemFirmwareDescriptor/SystemFirmwareDescriptor.inf [IA32, GCC5, RELEASE]

- Failed -
Build end time: 16:05:22, Feb.14 2018
Build total time: 00:00:18
```
I assume this is related to OpenSSL - at least some google results point to that. I'm using OpenSSL_1_1_0e as advised in documentation. Any pointers to solution appreciated.

BrianRichardsonIntel, 2018-02-14:
Pitor: Have you attempted to build in Ubuntu 16.04 LTS (versus Debian)? I've successfully built v0.97 without any need to modify the source. I made some shell scripts to automate setup on a 'clean' Ubuntu 16.04 LTS install (setup, fetch, build). You can grab the ZIP file and see if these work on your Debian config.

[lazy_edk2.zip](https://github.com/MinnowBoard-org/bugs-and-help/files/1725313/lazy_edk2.zip)

This might be a tools configuration issue rather than a source issue, especially since you're on an unstable branch of Debian. So the apt packages listed in the release notes might need to be adjusted for Debian jessie/sid. We tend to stick with Ubuntu LTS versions for build verification (stability).

If package or build instruction adjustments need to be made for Debian, we can update the TianoCore wiki to reflect these changes.

pietrushnic, 2018-02-14:
@BrianRichardsonIntel thank you for support. I was able to solve issue. As you suggested this was toolchain issue. GCC was in version 7 in Debian sid and version 5 is required. BTW error message is really criptic - it confuse even Google. Release notes of 0.97 not mention Ubuntu so instruction should work on any Linux.

I decided to put together Docker container to have separated build environment for edk2. For further reference you can find it in [3mdeb/edk2-docker](https://github.com/3mdeb/edk2-docker) there is also instruction how to build firmware fro MinnowBoard Turbot.

closed 2018-02-14 by pietrushnic

BrianRichardsonIntel, 2018-02-14:
Thanks for the follow-up. Glad you could resolve the problem.

The container is a great idea. I've had another engineer suggest this approach for development, so we might examine making these for future releases.

If you run other into firmware issues, please report them in the TianoCore Bugzilla. I suggest you open a feature request for GCC7 support, and a bug report against the MinnowBoard Max platform release notes regarding the ambiguity in build steps. https://github.com/tianocore/tianocore.github.io/wiki/Reporting-Issues

pietrushnic, 2018-02-14:
@BrianRichardsonIntel I will open feature request for GCC7. I just want to mention that release notes mention Ubuntu and GCC version:

```
Linux System Configuration:
  Ubuntu 16.04.2 64 bit
  GCC version: 5 or above
```

BrianRichardsonIntel, 2018-02-14:
Yes, the release notes should only refer to GCC 5. EDK II only has tools definitions for GCC 4.4-4.9 and GCC 5.
https://github.com/tianocore/edk2/blob/master/BaseTools/Conf/tools_def.template 
GCC 5 is recommended (smaller build size than GCC 4.x due to link time optimizations).

NikosMouzakitis, 2020-03-30:
> I was able to solve issue. As you suggested this was toolchain issue. GCC was in version 7 in Debian sid and version 5 is required

Could i please ask how did you managed to solve this? I am encountering the same problem with the same errors.

pietrushnic, 2020-03-30:
@NikosMouzakitis please us mentioned container [3mdeb/edk2](https://github.com/3mdeb/edk2-docker) everything is in README.md, if this would no work please describe what you tried and I will try to help.