MinnowBoard-org/bugs-and-help #93: GPIO voltage issue

Windows10010, 2018-02-17:
Here is the email to the support:

Good evening from Xi’an China
	I am recently doing some practice about libmraa and  GPIO tutorials with my new quad-core minnowboard turbot. But when i was going to blind an LED according to the tutorial archive, I found that the brightness of the LED was so low that i couldn’t sense it at first.
I though it may be the problem of the resistance of the LED. But when i tried to measure the voltage between the GND and GPIO 25 with my multimeter, i saw the value of 1.8 on the screen of the multimeter. I was kind of confusing because i clearly remembered the operating voltage of the minnowboard burbot is 3V3 instead of 1V8 and that is the main reason why i decided to choose it among too many dev boards. Is that something wrong with the hardware?? Or the system configuration?? DO i have to ship it back to the ADI for inspection??
If you need more info about the issue, please reply with a request. LOOKING FORWARD TO YOUR REPLY WITHOUT PATIENT.
											
yours
Liang Jiahe

Some grammar errors may occur. Please give me suggestions about them.

Windows10010, 2018-02-20:
Not only that. I also tested almost all the gpios on the board. Unfortunately, none of them are able to let a simple LED blink. And the voltage is 1V8 instead 3V3. That really freaked me out.

Mark-van-der-Pol, 2018-02-20:
You are correct - the expansion header GPIO are 3.3 v, and can certainly drive an LED to noticeable brightness. If you see 1.8v on the header that suggests a failure on the board of some sort, and I would recommend as a first resort to do a warranty replacement with your vendor.

Good luck!

|\\/|ark.

closed 2018-02-20 by Mark-van-der-Pol

Windows10010, 2018-02-20:
Thank you very much!!! I've been scared of this issue for a while.
And now I can certainly type the following into my brain`sudo shutdown now`!!! @Mark-van-der-Pol

jnavarro7, 2018-03-20:
Did you measure the GPIO pin voltage by itself or when driving the LED?  
Because when driving the LED the GPIO can't source the current needed by the LED and there is a voltage drop.   
Measure the voltage on the pin without any load, and to provide more current to the LED use a transistor as a switch with 3.3V directly from the power pin on the Minnow to the collector of the transistor.   
https://einhugur.com/blog/index.php/xojo-gpio/putting-the-gpio-pin-behind-transistor-to-get-more-current/

-Jose

BrianRichardsonIntel, 2018-03-21:
I never drive the LED directly from the GPIO on MinnowBoard Turbot (doesn't guarantee enough current). Jose's recommendation for a drive circuit is the best approach.