MinnowBoard-org/bugs-and-help #94: msata and mini-pice issue

Windows10010, 2018-03-10:
Hello there!

I just bought a lure from my local vendor. And the lure has two slots for mast and pcie.
I plugged in my intel msata sad into the lure while the board was running Ubuntu 16.04.3 but nothing happened. Because i also bought a usb to msatat adapter, so i used it to connect the ssd via USB3. It worked correctly.

I can ensure that there is nothing wrong with the lure. So it may be the problem of the board.
But there is still another possibility that i was using a 5v4A power to drive the board with a keyboard, a mouse and a wireless adapter. And the SSD required a 3.3v1A power source. So was the power not  enough? The board is dual-e and quad-core edition so it may require more power than the others.

If you guys have some suggestions, please reply me!
many thanks!

Windows10010, 2018-03-10:
When the SSD is connected via usb, i  added an external power source to the usb hub. And i think this may be the reason why the SSD didn't work well on the lure.

Windows10010, 2018-03-10:
The board was shorted out between the 5V and the GND once. So i am afraid of that this is the cause.

Windows10010, 2018-03-10:
After viewing the official support website of the install 525 SSD Series. I found that the SSD is not tested with any Linux OS and only officaially supported by Windows. So i tried to install the windows on it. But when i entered the installation page, windows also doesn't support it. So the problem cannot be about the compactibilities.

Will the pci bus be damaged?

If the lure mapping is correct, what caused the problem?

Windows10010, 2018-03-10:
ps : i have updated the firmware to the latest version.

Mark-van-der-Pol, 2018-04-13:
What is the exact lure you purchased? 

And causing a short on the board between 5v and GND may have stressed or blown one or more of the on board regulators, so that could definitely add to the problem.

There should be no issue with the OS's you have indicated at least seeing the SSD drive - but you don't say if running through the USB 3 port it successfully installs. 

|\\/|ark.

gmgallo, 2018-08-31:
Hi, I also have drive issues with the PCIe connection. I have a Turbot Dual Quad Core which includes a PCIe connector. Purchased a 64GB M.2 drive and connected to it. OS is Windows 10 Core, latest release. 
The problem is that I can't see the drive and I'm not even clear what de device name should be in the device tree. I wish I had DiskPart utility to list them but I don't, so, in a few words, I'm lost in the fog.

Windows10010, 2019-01-05:
I recently purchased a 16Gb sandisk msata SSD(SDSA4DH-016G sandisk). Fortunately, the board recognized it as a derive. So i install the ubuntu 18 LTS to the drive using my USB drive. After that, the disk cannot be detected by the bios unless i manually replug it into the slot and select reset in the main  bios menu so that it shows up next time i boot. I also tried many bios versions to do so, but only the latest firmware which i originally used recognize the drive. I must do the complex works over and over again just to get my board booted or just leave it running all the time. What can I do???

Windows10010, 2019-02-08:
Sorry. But to be true, I am very disappointed with this platform.....  I've spent most of my time on it trying to figure out the solutions to the strange bugs and issues instead of doing projects on it as intended.... With several strange issues unsolved, I finally gave up........

closed 2019-02-08 by Windows10010

gmgallo, 2019-02-08:
I'm not investing any more time in this badly supported board. It's a shame that we can't have a decent x86 based platform to develop with Windows IoT Core. I'm using exclusively Dragonboard 410C which is ARM based with a high end Snapdragon 410 CPU. It has a few issues of its own do to poor support from Qualcomm for Windows (BSP with broken serial ports) but I found a work around it using older BSPs which do work. For the most part it is a very stable platform.