MinnowBoard-org/bugs-and-help #96: Minnowboard Max Repair

gija1, 2018-04-05:
Hello everyone, 

I inadvertently supplied my MinnowMax with 12 V and now I don't get response of the system (like anticipated). The LED 'D2' also does not light up, so the system is not running. 

Are there any ways to repair the Board?

Thanks for your help

naikrovek, 2018-04-05:
I don't know, but after this many board revisions, it would seem that some overvoltage protection should be in place.  I would even consider it a hardware bug that overvoltage protection is not present.  

Sorry I can't answer you.

Mark-van-der-Pol, 2018-04-05:
The MinnowBoard Max indeed has no overvoltage protection. And later versions like the Turbot and Dual Ethernet Turbots do have overvoltage protection up to about 35v. Of course, that doesn't ease the pain of having blown the regulator on a MinnowBoard Max...  :-/

A repair is feasible, and there are notes [here ](http://elinux-minnowboard.elinux.narkive.com/MsF5io4x/repair-minnowboard-power-supply )and elsewhere on the web detailing how to do that. It does require tools and skills beyond a lot of people's capabilities. 

Good luck,
|\\/|ark

closed 2018-04-05 by Mark-van-der-Pol