MinnowBoard-org/bugs-and-help #99: what makes SMI/SMM loads on the CPU and how to enable/disable on minnowboard

seokheehan, 2018-05-14:
Hi,

I now use minnowboard turbot quad core board to develop realtime Linux system. and I find some big latency, about from 1.x ms to 2.x ms, on the CPU. I find it using hwlat tracer of Linux kernel and it tells that SMI or NMI can make some latency, which is not Linux kernel aware, on the CPU. Therefore, I would like to prove it, but I don't know BIOS, UEFI, and etc. well.
Could anyone support to find what makes SMI/SMM loads on the CPU on minnowboard and how to disable/enable them?

Thanks,

zwei4, 2018-05-15:
Two suspicious SMI sources at Linux run-time are:(1) Software SMI triggered by UEFI Variable run-time service; (2) SMI triggered by Digital Thermal Sensor on SoC.     Currently there is no user interfaces to disable them.    To make SMI triggered by Digital Thermal Sensor to be inactive,  could you try to cool down your board by fan.

pietrushnic, 2018-05-18:
@seokheehan did you considered to handle it by modding firmware either edk2-platform or coreboot?

seokheehan, 2018-05-23:
@zwei4 I guess so and I want to control any SMI signals if it is possible. However, I cannot find the right way yet.

seokheehan, 2018-05-23:
@pietrushnic I've used yocto-based linux, and I've just found grub-efi which I can disable usb-related SMI in. In fact, I'm not familiar with intel based bootloader or firmware, so I've handled edk2-platform or coreboot yet. I will find and try. Thanks a lot.

zwei4, 2018-05-28:
For enable/disable SMIs, please refer to section "30.2.5 SMI#/SCI Generation" of this datasheet:

https://www.intel.cn/content/www/cn/zh/embedded/products/bay-trail/atom-e3800-family-datasheet.html?wapkw=e3800+datasheet

bdelgado1995, 2019-06-24:
MSR 0x34 tells you the SMI counter so you could monitor this to see if it's increasing. If you see the latencies but the counter isn't incrementing, the source is not an SMI.