MinnowBoard-org/bugs-and-help #100: BSP for a commercial RTOS

charlieGrasses, 2018-05-19:
Hi all,
Im currently developing a minnow turbot dual-core BSP for a commercial RTOS. I have everything already up and running, except for the network. The board I have comes with Realtek pci device (at B3D0F0) and I need it to communicate its interrupt either via MSI or to the legacy 8259; I just need the interrupt to be raised and then I can configure my system accordingly.
I have tried different settings, but none have worked so far, and the interrupt is never raised. I am kinda new to PCI so I guess that is part of the problem ;-).
Could someone please provide some guidelines?
Thanks,

Mark-van-der-Pol, 2018-05-25:
Congratulations on submitting the 100th issue! PM me for a door prize for this.

Unfortunately, the Realtek ethernet chip was only on a single short-lived build of the Turbot, and was replaced with intel i211 and i210 on Dual and Quad core Turbot boards, in currently available stock. There are only a few Turbots in the wild with the Realtek chip, so it is unlikely worth your time to craft a BSP for this obsolete configuration.

These updated boards are typically silk screened Turbot 2 or Turbot B, and the network chip migrated from the back to the front of the board if you need to visually identify one.

I would recommend that you obtain a new board with the intel chips and continue developing the port there.

If that still doesn't play happy, feel free to reach out again.  

|\\/|ark

charlieGrasses, 2018-05-25:
Thanks Mark. This is the board my customer provided, so not sure if they want to get a newer one. I will let them now about this.
In the case that they still want to complete this development, what would be your recommendations/guidelines for raising the network interrupts as MSI?

Thanks again

charlieGrasses, 2018-06-05:
I have fixed the issue and network is working with this commercial RTOS.

closed 2018-06-05 by charlieGrasses

Mark-van-der-Pol, 2018-06-05:
Glad to hear that. 
I am curious what the breakthrough was? It might be something I can share in a tutorial or how-to.
|\\/|ark.

charlieGrasses, 2018-06-05:
Mark,
In the ISR, I was masking the interrupt and then sending the local APIC EOI. after some debugging I found out that the EOI is needed prior to masking the interrupt. I changed the logic a bit and the interrupt started being raised. I then adjusted everything else to use MSI.