MinnowBoard-org/bugs-and-help #101: Blue Screen of Death (BSOD) MinnowBoard Turbot Dual-Core with Windows 10 iOT Core

herlanderjorge, 2018-06-01:
Hello,

I bought 5 minnowboards MinnowBoard Turbot Dual-Core for a project and I'm having lots of issues with the board. There is something really wrong with this board and Windows 10 IOT Core version 10.0.17134.48.
I keep getting bluescreens when I leave apps running. It does not matter which app I ran, it always ends up on a BlueScreen. Sometimes it can run for 10 hours, but sometimes it crashes after 30 minutes. The only device that I have connected is a wifi dongle from TP-Link but I had BSOD with ethernet connection only. 

This is really affecting my project because I cannot stabilize any device. I have tried all different boards and the result is always the same.
is this a Windows IOT issue? Driver issue?

I have attached some crashdumps that I had and from from what I could understand is something related with:

```
DPC_WATCHDOG_VIOLATION (133)
The DPC watchdog detected a prolonged run time at an IRQL of DISPATCH_LEVEL
or above.
Arguments:
Arg1: 0000000000000001, The system cumulatively spent an extended period of time at
	DISPATCH_LEVEL or above. The offending component can usually be
	identified with a stack trace.
Arg2: 0000000000001e00, The watchdog period.
Arg3: fffff801008fe378, cast to nt!DPC_WATCHDOG_GLOBAL_TRIAGE_BLOCK, which contains
	additional information regarding the cumulative timeout
Arg4: 0000000000000000
```

[minnowboard_bsod_crashdumps.zip](https://github.com/MinnowBoard-org/bugs-and-help/files/2061713/minnowboard_bsod_crashdumps.zip)

BR, Herlander

herlanderjorge, 2018-06-01:
Hello,

That was my first check, it has the last one, the release version 0.97 from last September.

[image]

BR, Herlander

herlanderjorge, 2018-06-06:
Hello,

Any update on this? This is really affecting our project...

BR, Herlander

keerockl, 2018-06-06:
Sorry for the late response, here is what I got feedback from one of our engineers.

A quick check of the crash dump has yielded the following information:
A driver.. iaioi2c.sys  …  is an i2c driver created by intel… (Intel® Processor I/O Controller)
It is taking too much time in  an interrupt service routine.

If any driver spends too long in an interrupt service routine, you will get this bug check.
So, the first thing to try, is to make sure the IO drivers you are using are up to date, and support windows IoT core.

Can you make sure your Chipset drivers are up to date?
It is usually a good idea to install those before the other drivers..

herlanderjorge, 2018-06-08:
Thanks for the reply. 
Are you aware of this issue?

https://docs.microsoft.com/en-us/windows/iot-core/learn-about-hardware/pinmappings/pinmappingsmbm

I2C Issues
The MinnowBoard Max has a known issue with the I2C bus which causes communication problems with certain I2C devices. Normally, an I2C device will acknowledge its address during a bus request. However, under certain conditions this acknowledge fails to propagate back through the level shifters to the MBM, and as a result the CPU thinks the device did not respond and cancels the bus transaction. The issue seems to be related to the TXS0104E level shifters on the IO pins, which can trigger prematurely due to voltage spikes on the line. The current workaround is to insert a 100 ohm resistor in series with the I2C SCK line, which helps suppress spikes. Not all devices are affected, so this workaround is only required if you are having trouble getting a bus response. One device that is known to require this workaround is the HTU21D.

keerockl, 2018-06-13:
Thank you Herlander for your comment and information. We will further investigate internally and update changes as needed. For now please take the advice as mentioned in the link and please share the result to the community.
-Keerock

herlanderjorge, 2018-06-14:
Hello,

I did what you suggested on your previous post and built my own FFU with Intel latest drivers for I2c and all, and now everything is working fine. The board is stable and no more bluescreens.

BR, Herlander

closed 2018-06-19 by keerockl