MinnowBoard-org/bugs-and-help #104: Does the MinnowBoard Turbot Dual-Core require non-free software ?

Pomd0, 2018-06-27:
Hello,

I am considering buying a MinnowBoard Turbot Dual-Core.

Could anyone tell me if this motherboard requires any non-free software in order to work ? I am looking for a credit-card sized motherboard that can work with only free software (especially the GPU).

I also would like to know if the CPU and chipset initialization code is currently provided as a binary module.

Thanks in advance.

Regards,

gvancuts, 2018-06-27:
We have all drivers in open-source (inc. for the GPU, it's in the i915 and mesa stack).

You can find out more about the firmware here: https://firmware.intel.com/projects/minnowboard-max

Pomd0, 2018-06-27:
Thank you for your answer.

I see in the Intel's firmware information page that the Firmware Images are released as binaries and read that source code for the "Binary Object Modules" are not available. 

To me, this means that the minnowboard board requires proprietary software to boot up and work properly.

Could you confirm ?

Thank you for your help.

BrianRichardsonIntel, 2018-06-27:
Most open source firmware relies on a few initialization elements that are only available in binary format. MinnowBoard Max/Turbot uses a few binaries for init code that hasn’t been contributed to open source. Other Intel platforms use Intel FSP (intel.com/fsp) to support EDK II & coreboot platforms, which encapsulates those binaries in a standardized format to simplify integration.

Thanks … br
---
Brian Richardson, Senior Technical Marketing Engineer, Intel Software
brian.richardson@intel.com<mailto:brian.richardson@intel.com> -- @intel_brian (Twitter & WeChat)
https://software.intel.com/en-us/meet-the-developers/evangelists/team/brian-richardson

Mark-van-der-Pol, 2018-06-27:
Interesting question – what is open source software or free software?

In any current hardware platform running open source OS’s, there are myriad lines of code and software that are not free, not open source, proprietary and closed. This is the software that runs in the USB controllers, the network controllers, the power supply chip regulators, the WiFi module, the fan controllers, the display driver and pixel generators the bus controllers, the memory pipelines, etc, etc., etc.

Mostly, all that not open source, proprietary and closed code is burned into the devices themselves, and never sees the light of day, or is even of passing concern to most people. And, in some cases, the code is not burned into the device, but has to be boot-strapped into the device at poweron. These are the blobs that show up in boot blocks and firmware binaries. A lot of silicon vendors do it this way, as at least the blob can be revised and updated, and allows for all the benefits of field upgrading or patching ‘silicon’.  But it is unlikely that any of that software in those blobs will ever be open sourced. Too much secret sauce, too close to the silicon innovation and intellectual properties. And, for sure, a place where key-loggers and back-doors and denial-of-service and all sorts of nefarious dastardly plots can be hidden. And, by some accounts, have been hidden.

So, short of only running open source silicon with open source design logic and gate-code and micro-code, on top of which open source drivers support an open source operating system, the holy grail of completely open source hardware and open source software is curerntly unobtainable.

The efforts Intel is making to limit these islands of closed source code is laudable, and even if all the blobs eventually are opensourced, the micro-code buried and burned into the silicon will still be obscure, secret, protected and proprietary.

Which begs the question – what is open source or free software? The blobs described are free in that they are distributed to anyone who has a license for the silicon (ie they bought a chip) but they are not open source. Not really ‘free, either, as the price is factored into the chip cost, or into the purchase agreements to procure the chips.

Having said all that, there are movements afoot that attempt to create truly open sourced hardware and software – where ALL the design, logic, microcode, firmware, drivers, silicon black magic is in the open and unrestricted. If that is what you truly desire, then go support one of those endeavors. My sense is that it is still some way off getting wafer-up open source everything. But I hope we get there, someday.

That sounds very interesting, I didn't know about Intel FSP. It seems like that the Intel Galileo Gen 2 uses Intel FSP, but is has only 256MB of RAM - This pushes me to ask an off topic question (sorry) :

Could you please tell me what is the lastest x86 SBC that uses Intel FSP and has at leat 2GB of RAM ? And could you confirm that this SBC should be able to work with only free software ?

Thanks again for your help.

Pomd0, 2018-06-28:
Thank you for your answer.

> Which begs the question – what is open source or free software? The blobs described are free in that they are distributed to anyone who has a license for the silicon (ie they bought a chip) but they [...]

By free software, I was talking about the ability to read and modify the software. Free - as in Freedom, I am totally fine with software having a cost. I did not use the term Open Source, has such software and hardware usually depends one proprietary software.

Nevertheless, as you said, there are some cases where the software is burned on the hardware - And it seems like their is nothing much that can be done about that.

I take note that the Minnowboard can run on an Operating System without proprietary drivers installed, but that its hardware requires proprietary software to boot up.

Thanks again,

PS : For the record : My intention was to find out how much liberty the Minnowboard offered, as a consumer who loves IT.

closed 2018-06-28 by Pomd0

BrianRichardsonIntel, 2018-06-28:
Thanks for interest in the MinnowBoard. The intention of the project is to open as much of the hardware & firmware as possible, so people have the freedom to tinker and/or alter the design. We’ve worked on each iteration of product firmware to enable more features and reduce the number of binaries involved in the boot process. Getting to 100% source might not be possible, the goal is to only use binaries for silicon-specific functions when a BSD license can’t be applied (patent, business limitation, etc.).

Having open schematics and freely downloadable firmware source is a huge variation from past Intel platforms, and a trend that will hopefully continue. I’m glad we are able to produce something that provides more visibility into the hardware & firmware .

Thanks … br
---
Brian Richardson, Senior Technical Marketing Engineer, Intel Software
brian.richardson@intel.com<mailto:brian.richardson@intel.com> -- @intel_brian (Twitter & WeChat)
https://software.intel.com/en-us/meet-the-developers/evangelists/team/brian-richardson