MinnowBoard-org/bugs-and-help #108: TPM manufacturer Endorsement Credential Certificate

cchandler, 2018-08-22:
I've got a minnowboard running the tpm2 tools for just about every operation, but I can't seem to get the tpm2_getmanufec to work correctly. I call after creating a hardware endorsement key
`tpm2_getmanufec -V -G rsa -O ek_pub.out -E ECcert.bin https://ekop.intel.com/ekcertservice/`
but that service always returns 404. The only time I've gotten it to work correctly was with a hard coded test suite value directly from the tpm2 repo. Is Intel the correct manufacturer of record and are these credentials available?

Related
https://github.com/tpm2-software/tpm2-tools/blob/master/test/integration/tests/getmanufec.sh

https://manpages.debian.org/testing/tpm2-tools/tpm2_getmanufec.8.en.html

cchandler, 2018-08-31:
@BrianRichardsonIntel Any word from your colleagues or if I should be using a different approach here?

Documenting for completeness: Trying to verify that a given key must have originated from a proper TPM.

zwei4, 2018-09-02:
fTPM is not supported by the default BIOS on the board. You need to download the BIOS which supports fTPM from https://firmware.intel.com/projects/minnowboard-max. And make sure PTT is enabled in BIOS Setup.

cchandler, 2018-09-02:
Yes, it's already been updated + enabled. All the other TPM operations are working correctly.

cchandler, 2018-09-03:
I can try to put an email together but I'd like to leave a lot here in case anyone else finds this. There's supposed to be an endorsement credential or a platform certificate for verifying that a given EK came from a TPM. [Microsoft explains](https://docs.microsoft.com/en-us/windows/security/threat-protection/protect-high-value-assets-by-controlling-the-health-of-windows-10-based-devices) that Intel fTPM is supposed to be available at https://ekop.intel.com/ekcertservice . This works for the EK public key body in the tpm2tools repo https://github.com/tpm2-software/tpm2-tools/blob/master/test/integration/tests/getmanufec.sh#L62 .

If I run `./tpm2_createek -c - -G rsa -p ek_pub.out` and generate an EK from the hardware I get:
```
name: 000b6d849d63f9b68143d675b403d95f191bc8d4b2990d9c49a4e398609242256a5f
qualified name: 000b32d344ff14600770f5e9db6a0171313cb0457e7036220c21611137aef9ff30f3
name-alg:
  value: sha256
  raw: 0xb
attributes:
  value: fixedtpm|fixedparent|sensitivedataorigin|adminwithpolicy|restricted|decrypt
  raw: 0x300b2
type:
  value: rsa
  raw: 0x1
```
and these are all the correct template/parameter values from [the spec](https://trustedcomputinggroup.org/wp-content/uploads/Credential_Profile_EK_V2.0_R14_published.pdf) (see table 1: Default RSA template). But for some reason, the ekops service at Intel doesn't accept the base64 encoded name of this certificate when I send it through `tpm2_getmanufec`.

tpm2_readpublic + (bad) tpm2_getmanufect output
```
tpm2_readpublic -c 0x81000003 -o other.pub
name: 000b6d849d63f9b68143d675b403d95f191bc8d4b2990d9c49a4e398609242256a5f
qualified name: 000b32d344ff14600770f5e9db6a0171313cb0457e7036220c21611137aef9ff30f3
name-alg:
  value: sha256
  raw: 0xb
attributes:
  value: fixedtpm|fixedparent|sensitivedataorigin|adminwithpolicy|restricted|decrypt
  raw: 0x300b2
type:
  value: rsa
  raw: 0x1
rsa: bfe3bca0bed480b33d17f5ab7b748a477aff53ebcf46765c7a48c8bc4007e323b474d4a0b1162689b222e418e2fbb960852820366d73decb89baee67d4d2b7d2292eaf3c4db63435ed1b6454c5d198e1211bd51e13c31101a2ed0b06b8d14e2210324b02d44ad859c3ac0c3c914ebf274c00756f78a28c90002bb8f0e1368280bbe2879edae033b8157f83062a11370ab06cbed9c4b9b492660217e18fe0bc763bf9b8dbffe42c3e8a329d0a43290082d2ff762db4c7fabb1acd682d42a7ea293e268f37b320ef0757ad89ccdf03b3d45614471d2b6a7022338d39c530546774765175bc00a73f7335f38180181280975dc678eefa9cdbe4ed884a7b34426613
authorization policy: 837197674484b3f81a90cc8d46a5d724fd52d76e06520b64f2a1da1b331469aa

chris@minnow:~/Development/tpm2-tools/tools$ ./tpm2_getmanufec -V -O other.pub -E ECcert.bin 

https://ekop.intel.com/ekcertservice/
public-key-hash:
  sha256: B854ED72D602C67465B2F03A7235AED7D41250296C8ED764DBF67EB35C40044E
INFO on line: "294" in file: "tools/tpm2_getmanufec.c": Calculating the Base64Encode of the hash of the Endorsement Public Key:
INFO on line: "435" in file: "tools/tpm2_getmanufec.c": uFTtctYCxnRlsvA6cjWu19QSUClsjtdk2_Z-s1xABE4%3D
*   Trying 192.198.164.61...
* TCP_NODELAY set
* Connected to ekop.intel.com (192.198.164.61) port 443 (#0)
* ALPN, offering http/1.1
* successfully set certificate verify locations:
*   CAfile: /etc/ssl/certs/ca-certificates.crt
  CApath: /etc/ssl/certs
* SSL connection using TLSv1.2 / ECDHE-RSA-AES128-GCM-SHA256
* ALPN, server did not agree to a protocol
* Server certificate:
*  subject: C=US; postalCode=95630; ST=CA; L=Folsom; street=FM1-110; street=1900 Prairie City Rd.; O=Intel Corporation; OU=Hosted by Intel Corporation; OU=Multi-Domain SSL; CN=ekop.intel.com
*  start date: Sep  7 00:00:00 2017 GMT
*  expire date: Sep  7 23:59:59 2019 GMT
*  subjectAltName: host "ekop.intel.com" matched cert's "ekop.intel.com"
*  issuer: C=GB; ST=Greater Manchester; L=Salford; O=COMODO CA Limited; CN=COMODO RSA Organization Validation Secure Server CA
*  SSL certificate verify ok.
> GET /ekcertservice/uFTtctYCxnRlsvA6cjWu19QSUClsjtdk2_Z-s1xABE4%3D HTTP/1.1
Host: ekop.intel.com
Accept: */*

< HTTP/1.1 404 Not Found
< Cache-Control: no-cache
< Pragma: no-cache
< Content-Length: 21
< Content-Type: text/plain; charset=utf-8
< Expires: -1
< FWS: 4
< Date: Mon, 03 Sep 2018 21:36:13 GMT
[... 45 lines cut ...]

< HTTP/1.1 200 OK
< Cache-Control: no-cache
< Pragma: no-cache
< Content-Type: application/json; charset=utf-8
< Expires: -1
< FWS: 2
< Date: Mon, 03 Sep 2018 21:38:26 GMT
< Content-Length: 1659
< Set-Cookie: TS01da6661=01c5f9c67cacb8bdfa82350c6644563d27a5cbf6ff3a999d7022ba03ddb81ef733cee59bc14954ed6e7b47183e05700f40fa3c67bf; Path=/
<
* Connection #0 to host ekop.intel.com left intact
```

So this base64 EK public key came from the minnowboard and should be valid, but the ekops service only returns a 404. I have zero visibility into that service or why it isn't finding a result. I do, however, want to be able to verify that a given EK originated from the TPM and this appears to be the correct way to do it via the spec unless I'm missing something.

Base64 encoded version of my generated EK (decodes to raw bytes, not PEM).
```
AToAAQALAAMAsgAgg3GXZ0SEs/gakMyNRqXXJP1S124GUgtk8qHaGzMUaaoABgCAAEMAEAgAAAAA
AAEAv+O8oL7UgLM9F/Wre3SKR3r/U+vPRnZcekjIvEAH4yO0dNSgsRYmibIi5Bji+7lghSggNm1z
3suJuu5n1NK30ikurzxNtjQ17RtkVMXRmOEhG9UeE8MRAaLtCwa40U4iEDJLAtRK2FnDrAw8kU6/
J0wAdW94ooyQACu48OE2goC74oee2uAzuBV/gwYqETcKsGy+2cS5tJJmAhfhj+C8djv5uNv/5Cw+
ijKdCkMpAILS/3YttMf6uxrNaC1Cp+opPiaPN7Mg7wdXrYnM3wOz1FYURx0ranAiM405xTBUZ3R2
UXW8AKc/czXzgYAYEoCXXcZ47vqc2+TtiEp7NEJmEw==
```

idesai, 2018-09-04:
@cchandler the fTPM generation on minnow board (Intel (R) Baytrail SOC or Intel Atom® Z3700 Processor Series) is an older generation where EK certificates are not hosted by Intel. You can get this feature on newer generation fTPM on newer SOC Intel Atom® Processor E3900.

cchandler, 2018-09-04:
@idesai ugh. I see. Does that mean the previous generation EK certs are just totally unavailable? They aren’t even embedded somewhere in nvram on the device?

idesai, 2018-09-04:
@cchandler, EK certificates attest the endorsement primary seed or EPS that goes on to the TPM is known to the manufacturer, thereby guaranteeing the resulting EKpublic is on a TPM that was manufactured by the manufacturer. So there is a lot of backend infrastructure that has to vouch for the EPS along side programming the known EPS. This end to end feature set was enabled in the newer SOC. Having said that, for the EKpublic you see on the platform, you could potentially create a CSR and sign it with your keys. Not sure what your use case is.

cchandler, 2018-09-05:
@idesai We were hoping to extend an attestation system to an already-in-the-field set of hardware. Ideally the endorsement certificate would have established that we were communicating with the bona fide TPM. Sounds like that's not going to work now and we'll have to update the plan to go with key enrollment.
Unless there's a harddrive at the factory just sitting around with the certs on it and they just aren't hosted.

closed 2018-09-26 by cchandler