MinnowBoard-org/bugs-and-help #111: Need help on buying eval board

selvance, 2018-09-07:
I wanted to experiment on the firmware. Would like to know, Intel JTAG is necessary for the firmware development or I can go ahead with SF100 Dediprog?

zwei4, 2018-09-07:
Which board are you using? Minnowboard Turbot ?

SF100 Dediprog is for re-programming the firmware on SPI flash.
JTAG interface is  for code debug purpose.  JTAG does not work on Minnowbaord Turbot for some reason.  
Minnowboard Turbot UEFI firmware  provides alternative debug tool:  UEFI Source level debug over UART Serial Port.

selvance, 2018-09-10:
Yes, I'm planning to use Minnowboard Turbot.  
Thanks for the response.

zwei4, 2018-09-10:
SF100 is enough for getting started.

naikrovek, 2018-09-10:
I've reprogrammed the flash on minnowboards using a raspberry pi and some jumper wires.  The `flashrom` program on Linux can do everything you need if you already have a pi and some jumper wires.

selvance, 2018-09-10:
Thank you Zwei4, naikrovek and BrianRichardson.