MinnowBoard-org/bugs-and-help #113: Minnowboards freeze after a random amount of time

tchapi, 2018-10-01:
Hi

Similar to https://github.com/MinnowBoard-org/bugs-and-help/issues/64 or https://github.com/MinnowBoard-org/bugs-and-help/issues/57, I have 4 Turbot Quad boards that all freeze after a random amount of time, without any log, any preliminary sign, on different Ubuntu x64 versions (no modification of the OS whatsoever, just plain install). All the boards I have tested are on the latest firmware 0.98, are powered via a beefy 5V 6A power supply (tested different supplies to no avail), and have nothing connected to it except an HDMI screen.

I use industrial grade SD cards that work well.

I tried to contact support@minnowboard.org for a replacement but your mail server is broken (`Action: failed Status: 4.4.1 Diagnostic-Code: smtp; The recipient server did not accept our requests to connect. [minnowboard.com 52.87.45.42: timed out] [minnowboard.com 52.6.46.72: timed out]` )

What can I do to get a replacement for these boards ?
Thanks a lot

tchapi, 2018-10-19:
[image]

ftab, 2018-11-14:
I've got 3 minnowboards all doing the same thing, whether booting from SD cards or M.2 SATA SSDs. Disabling the multicore functionality seems to prevent it, but then of course you've paid for a quad core board that can't be a quad core board. And it's so slow when more than one thing's going on that it might as well have locked up.

MrMattBusby, 2018-11-14:
We have more than 10 units, all dual Eth, Quad core, every single one of them would freeze after XXX amount of time. Anywhere from 0.5min to 3days. Seemed to freeze more when less was running. 

We toyed with a few setting in the BIOS/OS but ultimately the ONLY consistent resolution we found was to not use the *recommended*  Ubuntu OS. CentOS/KDE has worked without a hitch. Buggy/disappointing...

tchapi, 2018-11-15:
... and nobody from Minnowboard responds ... ping @BrianOttaway ?

tchapi, 2018-11-15:
ping @jketreno too

BrianRichardsonIntel, 2018-11-15:
Have you contacted the board manufacturer (ADI/Silicom)?

umbaman, 2018-11-21:
I did contact them and I got no reply. But after upgrading to the latest firmware version (Release 1.00) the system seems to be working ok.
I haven't tested them thoroughly yet.
I have 2 such boards. Both have the same Ubuntu x64 version. I use SD card on one and SATA SSD on the other... I will report back if my systems get to crash again

naikrovek, 2018-11-21:
Do make sure (really make sure) that your power supply is up to the task, as well.  I have a 5V 4A power supply that shipped with the quad core turbot that I ordered and it worked for about 5 seconds.  It now delivers about 400mA at 4V.  

My point is not to trust the rating on the case.

If you have access to an electronic load, make sure your power supply can supply the power it claims it can supply.

This is more for future people who arrive here via Google than anyone specific in this thread.

tchapi, 2018-11-22:
Yes @naikrovek you are right. In my case (I am an electronics engineer and I do this kind of stuff for a living), I am sure of the rating of the supplies I have. The problems appears on 5.25V / 6A supplies that are very reliable otherwise (_I have concurrent boards that draw more than 5A and that work flawlessly on these supplies_). But it's a common culprit, thanks for pointing that out for people who would stumble on this issue.

@umbaman I'm glad to hear that ! I don't see anything related to our problem in the [release notes](https://firmware.intel.com/sites/default/files/minnowboard_max-rel_1_00_bin-releasenotes.txt) but I'm really interested to know if you experience any more crashes with this release

tchapi, 2018-11-30:
@umbaman unfortunately after having installed the 1.0 firmware and tested, I still have random freezes happening...

umbaman, 2018-12-05:
@tchapi  me too. Although they seem to be reduced, I still have them.
Still no reply from minnowboard

umbaman, 2018-12-07:
I left them running with the System Monitor in the foreground. Nothing unusual there...
Any minnowboard guys here?