MinnowBoard-org/bugs-and-help #114: Which driver should be installed?

WCL-GitHub, 2018-10-08:
I have bought MinnowBoard Turbot Dual-Core and installed Windows 10(x64) on it
but there are a lot yellow triangle exclamation mark in Device Manager
and the display resolution only have 800x600

There is only one driver for windows 10 in Intel website:
https://downloadcenter.intel.com/product/78477/Intel-Atom-Processor-E3826-1M-Cache-1-46-GHz-
"Intel® Embedded Drivers for Microsoft Windows® 10 IoT Core (32-bit and 64-bit) MR1"

but it can't insall on winsows 10(x64) ...

WCL-GitHub, 2018-10-11:
I have sent a mail to "support@minnowboard.org" two days ago and there is not reply...

zwei4, 2018-10-11:
For graphic driver, please use 

https://downloadcenter.intel.com/download/27417/-15-33-?product=85552

The readme only lists Atom Z3700 (Bay Trail–T) , but  it does work on Atom E3800 (Bay Trail–I) on Minnow Turbot .

WCL-GitHub, 2018-10-11:
I asked the same question here:
https://social.msdn.microsoft.com/Forums/windowsdesktop/en-US/6faa7747-ddf8-4e14-a998-df993f83cb22/which-driver-should-be-installed-on-minnowboard-turbot-dualcore?forum=WindowsIoT

I have updated through Windows Update and installed some driver includes graphic driver
but there are remaining two "other device" in device manager...

closed 2018-10-17 by WCL-GitHub