MinnowBoard-org/bugs-and-help #117: Windows 10 IoT Core not booting

soosterwaal, 2018-10-14:
I bought a Minnowboard Turbot Dual-Core last year and it worked fine. I want to pick up my work again, but the minnowboard is unable to boot the latest version of Windows 10 IoT Core, build 17763. I already updated the firmware of the minnowboard to the latest (0.99). I use a 8GB class 10 SD card, which should be enough for Windows IoT Core. After start up the minnowboard briefly shows `Boot failed. EFI Misc Device`.

What i did notices was that my Minnowboard has 32bit firmware, but the instructions on the Minnowboard site says it should be 64bits. Could this be the problem or am i missing something?

soosterwaal, 2018-10-14:
The firmware was the probleem

closed 2018-10-14 by soosterwaal

dynomation4, 2019-07-15:
How did you update the firmware from 32 to 64? I am having this exact issue but I am also having issues updating the firmware.

naikrovek, 2019-07-15:
If you are on a 32-bit firmware, you download the 64-bit firmware and the 32-bit firmware and use the 32-bit tool to flash the 64-bit firmware file.

dynomation4, 2019-07-15:
Thank you for the reply first and foremost. 

I did use the 32 bit flashing tool and a 64 bit bin file but after entering the commands I got no Echo reply (Error, success, etc)

See my new issue here for more detail: 

https://github.com/MinnowBoard-org/bugs-and-help/issues/148

dynomation4, 2019-07-15:
And now it's gotten worse. I seem to only be able to flash .FD files so I found version 1.00, .94 and m93 from the intel site. Well the 1.00 worked as it should but when I tried .94 it seemed to work until it rebooted and stayed stuck on a screen that has the logo for tianacore. Cant do anything from here except ctrl+atl+del and all that does is reboot the system