MinnowBoard-org/bugs-and-help #119: Help With I2C on Turbot

DuaneAugust, 2018-10-19:
I am trying to get an I2C sensor to work with my MinnowBoard Turbot (single ethernet; firmware version 0.99).  I am running Debian Linux (Stretch) installed on an SD card.  The sensor I am trying to use is this D-A converter:

https://www.adafruit.com/product/1083

The sensor address is 1001000 = 72 = 0x48.  Clock and data lines are connected to pins 13 and 15.

Here is some relevant information:

dadamia@Minnow-1:~$ lsmod | grep i2c
i2c_i801               24576  0
i2c_smbus              16384  1 i2c_i801
i2c_algo_bit           16384  2 igb,i915
i2c_designware_platform    16384  0
i2c_designware_core    20480  1 i2c_designware_platform
dadamia@Minnow-1:~$ 

dadamia@Minnow-1:~$ ls /sys/class/i2c-adapter
i2c-0  i2c-1  i2c-10  i2c-2  i2c-3  i2c-4  i2c-5  i2c-6  i2c-7  i2c-8  i2c-9
dadamia@Minnow-1:~$ ls /sys/class/i2c-adapter/i2c-0
delete_device  firmware_node    name        power      uevent
device         i2c-MSFT8000:00  new_device  subsystem
dadamia@Minnow-1:~$ ls /sys/class/i2c-adapter/i2c-1
delete_device  firmware_node  new_device  subsystem
device         name           power       uevent
dadamia@Minnow-1:~$ ls /sys/class/i2c-adapter/i2c-2
delete_device  firmware_node  new_device  subsystem
device         name           power       uevent
dadamia@Minnow-1:~$ ls /sys/class/i2c-adapter/i2c-3
delete_device  device  name  new_device  power  subsystem  uevent
dadamia@Minnow-1:~$ 

(Results for i2c-4 through i2c-10 are the same as for i2c-3.)

According to the tutorial at

https://minnowboard.org/tutorials/connecting-device-sensor-low-speed-expansion-lse-header

I should see a response from the "i2cdetect -r" command.

root@Minnow-1:/home/dadamia# i2cdetect -r 0
Error: Could not open file `/dev/i2c-0' or `/dev/i2c/0': No such file or directory

Same result for numbers 1 - 10.  There are no files of that sort in my /dev directory.  How are these files created?

What am I missing here?

naikrovek, 2018-10-19:
those are virtual files (they don't actually exist on disk) and only show up when the proper kernel drivers are loaded.

this is just a small bit of info for you -- I don't know how to fix that myself, aside from compiling a custom kernel with the required drivers included.  there may be a way to turn on those devices either in the BIOS or via configuration in Linux.

DuaneAugust, 2018-10-19:
It looks like I need a kernel module called i2c-bcm2708, which is not in the Debian repository.  There are about a million articles on the Internet explaining how to install this on a Raspberry Pi, but I can't find a version for Debian Linux on a MinnowBoard.

The source code is on GigHub here (second item):

https://github.com/bootc/linux/tree/8002705053f49235f3b530f3a998aa75dd1f33a7/drivers/i2c

I'm new to GitHub.  Do I just download the whole directory and run the makefile?  Does anyone have a pre-compiled i2c-bcm2708 for Debian Linux?

naikrovek, 2018-10-20:
That's the module you would use if you were using a raspberry pi.  So that's not what you need.

I don't know exactly what you need, but that's isn't it.

DuaneAugust, 2018-10-21:
You are right.  For Debian I needed to do the following.

1. Install packages libi2c-dev and i2c-tools.  Both are in the normal Debian repository.

2. Install the i2c-dev driver by running (as root) "modprobe i2c-dev".  This created i2c-0 through i2c-10 in /dev.

After doing that, the "i2cdetect -r 0" command ran and found my i2c device.

I have a lot more work to do to get my device to work, but at least I am through the first 3 steps of the tutorial at 

https://minnowboard.org/tutorials/connecting-device-sensor-low-speed-expansion-lse-header