MinnowBoard-org/bugs-and-help #123: issue: output only on serial, HDMI is off [turbot dual eth/dual core]

chiar-eu-insumi, 2018-11-04:
Summary:
I have made serial console work, after quite a bit of fiddling. But now I get no HDMI output at all although it was working before.

Context:
Main issue in setting up serial console wasn't the board itself, but my made-from-parts USB/serial cable: USB + FTDI breakout board + individual wires.
It turns out the FTDI breakout board had the pin markings reversed: what they referred to as TX was TX from Turbot's PoV - I was expectling it to be TX from FTDI's PoV. No big deal, but I took me a while to figure it out, and until doing so I messed around with everything else, including EFI config.

Details:
Now I don't get any HDMI output (TV says "no signal). The board accepts input either via serial or via USB keyboard.
Even after OS comes up (currently Alpine, but I have tried Ubuntu as well), HDMI is still off (both OSes are installed as command line only, no graphical interface).

I have obviously gone and loaded defaults from EFI menus, but it didn't make any difference.
I have even re-flashed firmware in the hope it'll reset it to factory state, but no success.

Again, it was working before, now it's not. I have tried swapping HDMI cables, using the other TV HDMI port without success.

Questions:
Is there something that I might have configured in EFI menu to disable console output on HDMI?
Can I get it back somehow?

It is possible that the board's HDMI somehow broke? I can't see how it'd happen just like that, as I didn't drop the board or anything, but it's one of the possibilities left..

zwei4, 2018-11-04:
As you have tried to re-flashed the EFI BIOS, this issue should not be caused by EFI BIOS setting.  It's possible that the HDMI receptacle is damaged?

chiar-eu-insumi, 2018-11-04:
HDMI receptacle looks to be in good order (no bends or anything) and everything passes visual inspection, including the circuits around the area.

The board is basically brand new (recently delivered) and I'm still toying with it before putting it to some use. I have barely used video signal for the EFI configuration (I modified quite a few things to be honest, for example downsizing the video memory allocation knowing I won't use graphical interface) and initial OS load.

I was hoping that there's a way to somehow persistently disable HDMI from the EFI menus and I have by mistake done it; the implication of this not being the case are worrying: means either physical or electrical damage.

Is there some linux utility that would allow me to probe HDMI port and it's state?

chiar-eu-insumi, 2018-11-10:
Hi,

It turns out that HDMI is working fine after all.

The ordering of the operations seems to make a difference: plug the HDMI cable before plugging the power cable. Just connecting HDMI after the board has been powered gets me into the "no signal" situation.
I have an old flat display Samsung TV, though (cca 2009), maybe this is the problem.
In any case, I'm happy there's no hardware fault. How does one close this issue here? Never used github before..

chiar-eu-insumi, 2018-12-10:
> A followup question to the issue @chiar-eu-insumi observed...
> Is there a way to force the presence of a display so the operating system is happy?
> I have found a bunch of options in the uncore configuration of the BIOS for LCD options, but none of the options seem to work. I was mainly trying to force HDMI to enable because I am using the board headless, however I use Teamviewer to see the desktop remotely. Without a physical display plugged into HDMI, the OS will not show a desktop (Ubuntu18).

I think that as long as you have an (active!) display controller the graphical mode should come up regardless if you have a monitor connected or not. Possibly your issue is somewhere else.