MinnowBoard-org/bugs-and-help #124: mSATA SSD or USB boot for Windows IoT Core

RGVGreatCoder, 2018-11-09:
Hello All,

I am building a prototype system that will make use of a single board computer running a UWP app on Windows 10 IoT Core. I am ready to test a MinnowBoard Turbot Dual Core board but before I purchase one, I would like to know the answers to my questions below:

1) Is it best to boot directly from a USB Stick or from an mSATA SSD?

2) Are there any particular brands of USB Sticks, mSATA SSD Extension Boards and mSATA SSD drives that work best with the MinnowBoard Turbot Dual Core board? 

3) Can you provide me with the exact extension boards and steps to setup Windows 10 IoT Core on an mSATA Internal SSD (Samsung 860 EVO, SUPTERBOOT-2) and have the MinnowBoard boot and operate through this SSD? 

4) What Bluetooth USB dongle would you recommend?

Thanks and excited to hear your reply soon! ... :)

zwei4, 2018-11-11:
For Minnowboard Max Lures for extension, please refer to https://www.elinux.org/Minnowboard:MaxLures;

For Minnowboard Max OS installation, please could refer to https://minnowboard.org/tutorials

RGVGreatCoder, 2018-11-11:
I just reviewed your second link which covers steps to install Windows 10 IoT Core on a microSD Card. Sorry. That does not answer my question. I am asking how to install and boot from an mSATA SSD drive. Do you have a link for my questions?  :)

chiar-eu-insumi, 2018-12-10:
While not having tried it myself, I don't see why the UEFI firmware can't do it.
I have a dual-ethernet Turbot (these have a B-keyed M.2 slot instead of the SATA port) and the firmware (it's the same across dual/quad-core single/dual-ethernet) could see my M.2 2280 32GB SSD at boot time (in addition to the microSD card, which I ended up installing Linux on).

It'll possibly be up to you to create the EFI partition, though (on Linux you have to roll your own, don't know how Windows does it).

BrianRichardsonIntel, 2018-12-10:
You can use an M.2 device on the MinnowBoard Turbot, but not all versions of the board have an M.2 slot. As noted above, the dual-ethernet boards have an M.2 slot. I am using one at home with openSUSE. 

The original Max & Turbot boards support expansion cards ("lures"). The Silverjaw & Flotsam lures add M.2 slots via the high-speed expansion bus. https://minnowboard.org/add-ons

M.2, mSATA & SATA devices should appear as normal SATA devices to the OS & firmware boot menus.