MinnowBoard-org/bugs-and-help #125: 4 Port USB 3.0 Hub for MinnowBoard Turbot Dual Core

RGVGreatCoder, 2018-11-09:
Hello All,

I am building a prototype system that will make use of a single board computer running a UWP app on Windows 10 IoT Core. I am ready to test a MinnowBoard Turbot Dual Core board but before I purchase one, I would like to know the answers to my questions below:

1) If a 4 port USB 3.0 Hub is connected to the USB 3.0 port of a MinnowBoard Turbot Dual Core, will the board running Windows 10 IoT Core be able to interact with devices connected to the hub, like an ASUS touchscreen, LIfecam webcam, an UHF RFID Reader using UART communication among other devices?

2) What Bluetooth USB dongle would you recommend?

Thanks and excited to hear your reply soon! ... :)

TheoRettisch, 2019-01-10:
Will be hard to find someone who uses exactly the same devices with the MinnowBoard.

Myself im using a powered 4 port USB3.0 hub with Realtek RTS 5411 chipset. No issues so far with a touchscreen, lte modem, MCU and hdd connected.

As for the bluetooth dongle just get one which has good Windows support.

naikrovek, 2019-01-10:
1) yes, provided that the USB hub is supported.  I've used Anker powered USB hubs without issue.  I can't speak to the devices you've mentioned, but USB HID stuff like touch screens have worked for me previously.

2) If I recall, this one has worked for me: https://www.amazon.com/dp/B009ZIILLI 

Note that Microsoft has a list of tested hardware, here: https://docs.microsoft.com/en-us/windows/iot-core/learn-about-hardware/hardwarecompatlist