MinnowBoard-org/bugs-and-help #127: Driver debug/deployment via Visual Studio 2017 timeout

jkanta, 2018-12-12:
I've spent a couple of days attempting different ideas in deploying a kernel-mode driver to a Minnowboard Turbot via Visual Studio 2017.  Many of the dialogs on the Microsoft page are just wrong, indicating that after the WDK is installed there should be a selection for Device Type.  Secure Boot is Disabled and I've touched nothing in the BIOS.  I did get Windows IoT running successfully, and I've tried both a company network and a private network (192.168.x.y).  I can connect via PowerShell, but I just don't understand the provisioning and other things that need to happen before Visual Studio can use the Minnowboard running Windows IoT as a device.  All versions of SDK and WDK are current as of this posting.

Any help would be greatly appreciated.

naikrovek, 2018-12-13:
see the "known issues" section of the release notes.  

https://docs.microsoft.com/en-us/windows/iot-core/release-notes/currentcommercial

jkanta, 2018-12-13:
I once found something about manually provisioning the IoT device for Visual Studio deployment, is that a possible workaround?  Is there any known workaround, at this time?