MinnowBoard-org/bugs-and-help #130: Turn Fan On/Off Turbot Dual Ethernet Quad-Core

atclaus, 2018-12-24:
Is there a way to control the fan with software based on temperature? My use is in my entertainment cabinet and the fan always on is more noise than I would like. I unplugged it for now, but am running around 70C without a load and would like to drop it.

TheoRettisch, 2019-01-10:
To my knowledge there is no way to modify the voltage or turn the pin on/off in software. 

One solution could be to swap the fan for a bigger/quieter 5V! model which could be mounted in the case instead on top of the heatsink. 

Depending on ambient temperatures you also might be able to use a passive cooler like this one:

Enzotech CNB-S1
http://www.enzotechnology.com/cnb_s1.htm

No guarantee this will fit as if haven't tried it myself yet. But looks promising.

warthog9, 2020-08-02:
The pins that the fan use are not attached to PWM or anything, and are on when the system is on