MinnowBoard-org/bugs-and-help #131: Legacy Mode/Boot?

TR1PL3D, 2019-01-06:
Hi,

I dont see an option to enable Legacy Mode within the BIOS. After further reading is would appear that my MBT-4220 (Dual Ethernet/Quad Core) is UEFI only. I have software I need to install that does not currently support UEFI.

I have read snippets of information about the possibility of using different firmware to allow for Legacy Mode.

Is there firmware available to allow Legacy Mode for the MBT-4220?

Thanks

naiyining, 2019-02-18:
I have the same problem.
Can any one who have Legacy Bios send me email naiyining@live.com?

Thanks.

naiyining, 2019-02-18:
I want to config(EEUPDATE) the network  IC I211 in Legacy Dos mode,but the board can't run in dos.

asa1365, 2019-09-07:
you can use Coreboot with seabios  and compile it for your board . it works well.

TR1PL3D, 2020-06-21:
> you can use Coreboot with seabios and compiling it for your board . it works well.

Interesting, is there a guide on how to do such a thing? I have no experience with compiling things like this.

Thanks

naiyining, 2020-06-22:
also interesting too...