MinnowBoard-org/bugs-and-help #137: Reset BIOS defaults on Turbot Dual Ethernet

graspomatic, 2019-03-12:
In the course of trying to debug a USB issue, I disabled XHCI and this has caused me to be unable to use any USB device. Now I can't boot into the BIOS and re-enable XHCI because the keyboard doesn't work. 

I thought I could reset the BIOS by removing the onboard battery, but after removing it for 10 minutes the problem persists. How can I reset the BIOS without USB support?

naikrovek, 2019-03-12:
You will need to re-flash the BIOS via the pins on the board.  If you have a Raspberry Pi you can use that to do it from within Linux using a tool called `flashrom` after things are wired up.

graspomatic, 2019-03-12:
Is there a guide for this somewhere? Which pins?

naikrovek, 2019-03-12:
https://3mdeb.com/firmware/flashing-minnowboard-turbot-with-raspberry-pi-zero-w/

BrianRichardsonIntel, 2019-03-12:
https://firmware.intel.com/sites/default/files/Flashing_MinnowBoard_MAX_with_Dediprog_SF100_in_Linux.pdf

https://firmware.intel.com/sites/default/files/Flashing_MinnowBoard_MAX_with_Dediprog_SF100_in-Windows.pdf

graspomatic, 2019-03-12:
For future reference, I managed to solve the issue without needing to reflash the UEFI. You just need to short jumper J7, then hit the delete key repeatedly on boot.

http://uchan.hateblo.jp/entry/2018/01/09/075230

Alexander-69, 2020-07-12:
hello i am having a similar problem, can someone help?

Alexander-69, 2020-07-22:
nvm