MinnowBoard-org/bugs-and-help #143: Supporting LVDS panel on eDP1 on Minnowboard?

rksaripalli, 2019-05-31:
Hi, I would like to support a custom LVDS panel on the eDP1 interface (DDI1_XXX signals on the SoC) on the Minnowboard.

I realize this port is removed on the Minnowboard and only HDMI is supported on the DDI0_XXX signals.

We have a separate board that is roughly modeled on the Minnowboard but has different schematics. We are able to compile and load the UEFI FW from Minnowboard on our board (with some patches for memory controller).

I have tried modifying various options in the UEFI FW menu and choosing IGD boot display to all the different options (DSI Port B, DSI Port C, eDP Port) but nothing works. The HDMI works fine but I really need to get the LVDS panel working. What do I need to do?

Thanks,
RK

BrianRichardsonIntel, 2019-06-03:
You may want to reach out to the EDK II mailing list for help modifying the firmware for LVDS.

https://github.com/tianocore/tianocore.github.io/wiki/Mailing-Lists

Thanks … br
---
Brian Richardson -- Director of Firmware Ecosystem Engagements
brian.richardson@intel.com<mailto:brian.richardson@intel.com> -- @intel_brian<https://twitter.com/Intel_Brian>
Firmware Ecosystem & Business Development, Firmware Integration & Development (FID)
https://software.intel.com/en-us/meet-the-developers/evangelists/team/brian-richardson

naiyining, 2019-06-06:
Is anyone has the available BIOS firmware for LVDS?
I have the same problem.Thx.