MinnowBoard-org/bugs-and-help #144: build BIOS, encounter ACPI BIOS ERROR problem

naiyining, 2019-06-16:
I build BIOS successful follow this link https://firmware.intel.com/sites/default/files/minnowboard_max-rel_0_99-releasenotes.txt, but when hardware powerup, encounter the ACPI BIOS ERROR problem.
can anyone help?

BrianRichardsonIntel, 2019-06-17:
Are you compiling from source or using one of the pre-built binary firmware images?

If you are compiling from source, please try one of the pre-built images and see if you get different behavior.

Thanks … br
---
Brian Richardson -- Director of Firmware Ecosystem Engagements
brian.richardson@intel.com<mailto:brian.richardson@intel.com> -- @intel_brian<https://twitter.com/Intel_Brian>
Firmware Ecosystem & Business Development, Firmware Integration & Development (FID)
https://software.intel.com/en-us/meet-the-developers/evangelists/team/brian-richardson

naiyining, 2019-06-17:
if I use pre-built binary firmware images, the MinnowBoard works well.
if use compiling from source images, the MinnowBoard  encounter the ACPI BIOS ERROR problem.
32bit or 64bit have the same problem.

naiyining, 2019-06-17:
https://firmware.intel.com/sites/default/files/minnowboard_max-rel_1_00-releasenotes.txt
I build BIOS successful follow this link, and it also have the same problem.
can anyone help? thanks.

BrianRichardsonIntel, 2019-06-17:
What is your build environment? (OS, Compiler)

Please check the installation of the ASL compiler per release notes:

3. Install the iasl compiler by downloading iasl-win-20160527.zip from the following location:
https://acpica.org/downloads/
and place the unzipped content ("iasl.exe") into the directory "C:\ASL" on your local hard drive (create the folder "C:\ASL" if it does not exist).

Note: you will get better answers related to EDK II build problems on the TianoCore groups.io list or IRC channel.

https://github.com/tianocore/tianocore.github.io/wiki/Mailing-Lists

Thanks … br
---
Brian Richardson -- Director of Firmware Ecosystem Engagements
brian.richardson@intel.com<mailto:brian.richardson@intel.com> -- @intel_brian<https://twitter.com/Intel_Brian>
Firmware Ecosystem & Business Development, Firmware Integration & Development (FID)
https://software.intel.com/en-us/meet-the-developers/evangelists/team/brian-richardson

naiyining, 2019-06-20:
it work ok after update the tools.
thanks.

closed 2019-06-20 by naiyining