MinnowBoard-org/bugs-and-help #150: Update firmware on MT25QL128

renaud-dk, 2019-08-25:
Hi,

We have designed a CPU board around the Intel Atom E3825 based on the Minnowboard-Turbot Dual Ethernet.
We have trouble to update the firmware with the FirmwareUpdate.efi application.
We are able to update from an Firmware build in debug mode to a Firmware build in release mode. Once the firmware in release mode is burned in the flash the FirmwareUpdate.efi will never be able to update the firmware in the flash. 

We have put a spy on the spi bus between the flash and the CPU and we have observed the following :
- In debug mode when a 4kb block of the flash needs to be updated an erase command (0x20) is issued before the page program command (0x02).
- In release mode when a 4kb block of the flash needs to be updated in place of the erase command (0x20) a Fast Read Dual I/O (0xBB) is issued before the page program command (0x02). Thus when the FirmwareUpdate.efi verifies the data it says that update fails and it exit.  

Our flash is a MT25QL128 which is correctly identified by the Firmware (Vendor Id 0x20, Device Id 0 0xBA Device Id 1 0x18).

We have build our sources tree following the procedure https://firmware.intel.com/sites/default/files/minnowboard_max-rel_1_01-releasenotes.txt.

We have also tried prebuilt firmware images for the minnnowboard in debug and release mode and observe the same problems.

Any hint of how to resolve this would be appreciated ?

BrianRichardsonIntel, 2019-08-26:
> Hi,
>
> Here an update.
>
> Today we replace the flash on our Minnowboard with a MT25QL128 and flash it with the firmware MNW2MAX1.X64.0101.D01.1908071825.bin though USB/SPI adapter.
> Than performed the update to MNW2MAX1.X64.0101.R01.1908071815.bin through the FirmwareUpdateX64.efi that was successfull.
> We tried to update the firmware back to MNW2MAX1.X64.0101.D01.1908071825.bin throug the FirmwareUpdateX64.efi that fails. When sniffing the SPI bus between the flash and the CPU we observe the same behavior as my previous post, the Fast Read Dual I/O (0xBB) command is issued in place of Erase (0x20) command.
>
> So the issue seems to be related to handling of the flash in the firmware.

You must update the firmware to work with the larger flash part. This is the likely reason he firmware update utility is failing. The pre-built BIN file is compiled to work with the default SPI part.

Thanks … br
---
Brian Richardson -- Director of Firmware Ecosystem Engagements
Firmware Ecosystem & Business Development, Firmware Integration & Development (FID)
[AllyNation-emailSignature]<goto/allynation>

Thank you

renaud-dk, 2019-08-26:
Hi,

Here an update. 

Today we replace the flash on our Minnowboard with a MT25QL128 and flash it with the firmware MNW2MAX1.X64.0101.D01.1908071825.bin though USB/SPI adapter. 
Than performed the update to MNW2MAX1.X64.0101.R01.1908071815.bin through the FirmwareUpdateX64.efi that was successfull.
We tried to update the firmware back to MNW2MAX1.X64.0101.D01.1908071825.bin throug the FirmwareUpdateX64.efi that fails. When sniffing the SPI bus between the flash and the CPU we observe the same behavior as my previous post, the Fast Read Dual I/O (0xBB) command is issued in place of Erase (0x20) command.

So the issue seems to be related to handling of the flash in the firmware.

Any hint of how to resolve this would be appreciated ?

Thank you

renaud-dk, 2019-08-26:
Hello Brian,

Thank you for your comment. I can understand your point about updating the firmware to use larger parts. However what's I don't get is why using a firmware compiled in debug mode is able to perform the update and a firmware compiled in release mode (from same source tree) is not able to perform the update. 

Thank you

BrianRichardsonIntel, 2019-08-26:
I agree that’s odd. I recommend asking this question on one of the TianoCore mailing lists, where the EDK II developers hang out. https://github.com/tianocore/tianocore.github.io/wiki/Mailing-Lists

Thanks … br

renaud-dk, 2019-08-26:
Thanks again for the comment. I'm closing the issue here and will post it on Tianocore mailing list.

Best regards

closed 2019-08-26 by renaud-dk